DexterInd/GoPiGo3 #64: No module named mock_package

angelarw, 2017-07-20:
the latest commit is causing the easygopigo3 to not work. 
"  File "build/bdist.linux-armv7l/egg/easygopigo3.py", line 2251, in <module>
ImportError: No module named mock_package"

this line here: references "mock_package" which is not defined elsewhere
https://github.com/DexterInd/GoPiGo3/blame/master/Software/Python/easygopigo3.py#L2251

DexterInd, 2017-07-20:
Thanks for the alert here @angelarw we will look at it immediately!

RobertLucian, 2017-07-20:
@angelarw it's now working.

closed 2017-07-20 by RobertLucian

guuskoning, 2017-09-20:
Ok, but what should I do now? I've updated the Gopigo 3 firmware but that didn't help.

CleoQc, 2017-09-21:
Did you update the GoPiGo3 library on your pi?

guuskoning, 2017-09-21:
Yes, I've followed the tutorials but when I launch the Scracth program I don't get the message "Remote sensor connections enabled". I don't get any errors but the program doesn't work.

CleoQc, 2017-09-21:
For now, Scratch will only work if you have Raspbian for Robots. Is that the case with your setup?

guuskoning, 2017-09-21:
Yes, I have Raspbian for Robots