DexterInd/GoPiGo3 #163: DistanceSensor maximum range

christianrauch, 2018-03-18:
The documentation of the DistanceSensor states that the minimum and maximum range is 5mm and 8000mm (8m) respectively.
https://github.com/DexterInd/GoPiGo3/blob/f484e09ab5154efd02b2df5d755b4f33e458c2b5/Software/Python/easygopigo3.py#L2702

The highest readings that I get with the `easy_Distance_Sensor.py` example are 3m (3000mm).

Which are the correct ranges for the DistanceSensor? Does the sensor need to be configured to allow higher ranges?

christianrauch, 2018-03-18:
I just saw that when using the `DistanceSensor` from `di_sensors`, I get readings in the maximum range of 8000mm, also with a higher update rate.
Is it then recommended to use the sensor instances directly than via the `EasyGoPiGo3` class?

mattallen37, 2018-03-19:
With the default configuration, the usable sensor range is approximately 5mm to 2000mm. By changing the configuration for greater accuracy, the usable range is decreased from roughly 2000mm down to e.g. 1000mm.

The sensor value register is 13 bits, and will return MAX (8191) if the sensor doesn't see an object in range.

The sensor value returned sometimes falsely indicates that there is no object detected. easygopigo3.py read_mm() polls the sensor multiple times to ensure that a false 8191 will not be returned. read_mm will only return "out of range" value if three consecutive readings all agree. This slows down reading the sensor, but increases the accuracy.

christianrauch, 2018-03-19:
Ok, I see. I haven't dug to deep in the implementation details for `DistanceSensor` (VL53L0X). As far as I can tell, the `EasyGoPiGo3` class is just reusing the `DistanceSensor` class from the `di_sensors` package, right? I.e. `EasyGoPiGo3.init_distance_sensor()` is instantiating a `DistanceSensor` from the `easygopigo3` library, which in turn is instantiating a `DistanceSensor` from the `di_sensors` library. The only difference is then which methods they use for acquiring the data, i.e. via `easygopigo3` it is using `read_range_single()`, and via `di_sensors`, you can directly use either `read_range_single()` or `read_range_continuous()`.

What are the implications of using these two modes other than having a higher update rate? E.g. the documentation says that `read_range_continuous` is faster and more accurate. So, would you recommend to always use this method for acquiring the sensor data (i.e. is there a downside of using `read_range_continuous`)?

Sorry for asking all these questions about the sensor ranges. I just want to make sure, that I am using the devices in a proper way.

CleoQc, 2018-03-19:
There are more significant changes i you look into the develop branch, mainly the use of mutex for multithread multiprocess safety. Other than that, it's just a simplified wrapper of the basic DistanceSensor class.

mattallen37, 2018-03-19:
The sensor in continuous mode will use slightly more power, as the sensor will continuously take readings. You can read a value from the sensor instantly in continuous mode, as it will just return the most recent value.

In single shot mode, the sensor only takes a reading when you request one (i.e. when you call read_range_single()). There is therefore a small delay, due to telling the sensor to get a reading, waiting for it to complete, and then finally reading the value.

You can use either mode. For a large majority of applications there shouldn't be a noticeable difference.

closed 2018-03-21 by christianrauch