DexterInd/GoPiGo3 #164: Valid servo range unclear from documentation

christianrauch, 2018-03-19:
In the `gopigo3` library, the range of servo motions is given with 0 to 16666 us (microseconds): https://github.com/DexterInd/GoPiGo3/blob/f484e09ab5154efd02b2df5d755b4f33e458c2b5/Software/Python/gopigo3.py#L445
But when I set anything lower than 500us or larger than 5000us, the servo hits the lower or upper limit.

In the `easygopigo3` library, the servo limits are given with 575 to 24250us:
https://github.com/DexterInd/GoPiGo3/blob/f484e09ab5154efd02b2df5d755b4f33e458c2b5/Software/Python/easygopigo3.py#L2585-L2586

Which are the true limits for setting the servo position?
Are there different kind of servos? If not, the documentation needs to be updated to contain the true valid ranges for servo motion.

mattallen37, 2018-03-19:
The servo control PWM signal is running at 60Hz. By setting the uS pulse width to somewhere from 0 through 16666 (1,000,000uS / 60Hz), you can set the PWM duty cycle from 0% through 100%.

For servos, 1500 uS is typically considered to be "center" or "neutral". Depending on the servo, the usable range could be 1000-2000uS (1000uS range), 250-2750 (2500uS range), or anything else. Additionally, depending on the manufacturing tolerances etc., the usable range could be something like 750-2500 (not centered at 1500). Even two "identical" servos will be slightly different, due to manufacturing tolerances and the fact that the servos are actually analog (internally they use a POT rather than an encoder for position feedback).

From a software and GPG3 standpoint, the available range is 0-16666. From a servo design and mechanical perspective, the usable range could be anything within 0-16666 (see above).

Note that by setting the servo control PWM signal to 0uS, you are essentially "turning off" the servo control signal, which most (if not all) servos will interpret as "float" (allow the servo to be rotated physically, as if it was unplugged).

The comment in easygopigo3.py on line 2586 has a typo. That "24250" should be "2425". @CleoQc

christianrauch, 2018-03-19:
Ok, thanks for the explanation. I think most servos actually have a usable range of 1000 to 2000us, since these is the timings that most RC transmitter use (e.g. for RC aircrafts).

If I want to provide the user with a interface where they set the servo position in a range 0.0-1.0 (i.e. 0.5 is the centre position), is it reasonable to assume 575-2425 us as a reasonable range that works for most servos that people use for the GoPiGo3?

mattallen37, 2018-03-19:
Most servos (including the GoPiGo Servo) have about 180 degree range. easygopigo3.py rotate_servo() uses degrees (0 - 180 degrees), and scales it to the range of 575-2425 uS (see [here](https://github.com/DexterInd/GoPiGo3/blob/f484e09ab5154efd02b2df5d755b4f33e458c2b5/Software/Python/easygopigo3.py#L2597-L2607)). This works okay for most applications with the GoPiGo Servo, but may be less than optimal for other servos (each one is different).

closed 2018-03-20 by christianrauch