DexterInd/GoPiGo3 #192: Is there a way to provide the odometry in the as a function?

wilianson, 2018-06-12:
Hi, I am trying to test  in Matlab some controllers for Mobile Robots, I am using a sockets communication between Python and Matlab, but there is a problem when I try to compute the odometry in Matlab, I was wondering if it is possible to me modify the source code in order to make the odometry and provide the position of the robot in centimeters o meters, and its orientation, What steps Should I follow?.
Thanks

slowrunner, 2019-03-20:
The GoPiGo3 does not maintain a position estimate, nor an orientation estimate for an external processor to interface with.  

The request should not be considered an "issue" with the GoPiGo3 implementation.  

Suggest closing the issue.  Requester can pursue such questions in the support forum https://forum.dexterindustries.com/

RobertLucian, 2019-03-20:
I have to agree with that. Closing this down.

closed 2019-03-20 by RobertLucian