DexterInd/GoPiGo3 #251: Error translating line follower Bloxter Code to Python

Auroric, 2019-07-09:
When using the line follower blocks, "line follower (red) sees" and "line follower (black) sees", the equivalent python code has a syntax error, causing the code to fail to run. If instead one checks if the line position is equal to a string using the "line_follower values" and the equals and text blocks, the issue is not found. I believe this occurs because, when using the "line follower (red) sees" and "line follower (black) sees" code blocks, the colon that ends the if statement is on the next line.  I have included images of both functioning bloxter code and non-functioning code (where the python equivalents are almost identical, except the colon as noted previously). 

[image]

[image]

[image]

CleoQc, 2019-07-09:
@Auroric 
You are indeed right. Thanks for reporting the issue and debugging it for us. Fix is incoming.

CleoQc, 2019-07-11:
There is an update for DexterOS available now:
https://www.dexterindustries.com/dexteros/update-dexteros/

(it took a while to post cause we experienced a website issue, the update itself was done pretty quickly)

closed 2019-07-11 by CleoQc