DexterInd/GoPiGo3 #269: Is GoPiGo3 fully compatible with Raspberry Pi 4?

jjogeva, 2019-11-19:
I can see some changes in the repository related to RPi 4. But I am unable to find a statement that would confirm or deny the support for RPi 4. I have also looked at Dexter Industries and done the old Googling trick.

Has this been fully tested?

CleoQc, 2019-11-20:
We've run tests and are not seeing any issues. The official statement is coming out this week.  
We do not recommend the Pi4 over the Pi3B+ because of battery usage. Unless you're doing machine learning, you're better off with the Pi3B, or Pi3B+.

closed 2019-11-20 by CleoQc

jjogeva, 2019-11-28:
Thank you for the quick response I have yet to pump into this statement - I'll try to add a link here as well so that when someone else wonders here they have an affirmative answer.

I am just going to add another link here for those more interested in the topic:
https://forum.dexterindustries.com/t/wifi-light-off-raspberry-pi4/6678/4

jjogeva, 2020-07-31:
Dexter now seams to have the support.
I'll add the link to the official post here as github ranks high in search results:
https://www.dexterindustries.com/raspbian-buster-and-the-raspberry-pi-4/