DexterInd/GoPiGo3 #270: test_heavy_mutex_2.py  TabError: inconsistent use of tabs and spaces

slowrunner, 2020-01-05:
[test_heavy_mutex_2.py](https://github.com/DexterInd/GoPiGo3/blob/master/Software/Python/tests/test_heavy_mutex_2.py) has an errant tab:
```
pi@Carl:~/Dexter/GoPiGo3/Software/Python/tests $ python3 test_heavy_mutex_2.py 
  File "test_heavy_mutex_2.py", line 20
    mutex.release()
                  ^
TabError: inconsistent use of tabs and spaces in indentation
```

slowrunner, 2020-01-05:
Same error in both [test_distance_sensor.py](https://github.com/DexterInd/GoPiGo3/blob/master/Software/Python/tests/test_distance_sensor.py) and [test_distance_sensor_2.py](https://github.com/DexterInd/GoPiGo3/blob/master/Software/Python/tests/test_distance_sensor_2.py)

CleoQc, 2020-01-06:
Done, thanks for reporting it!

closed 2020-01-06 by CleoQc