DexterInd/GoPiGo3 #300: Updating GoPiGo3 on Ubuntu 20.04.2

slowrunner, 2021-07-18:
My ROS2 GoPiGo3 is running well, but needs the latest gopigo3.py with the encoder ticks handling, so I tried:
https://github.com/DexterInd/GoPiGo3/tree/master/Install#subsequent-updates  
but fails with (full console log below):
`Executable "pip2" couldn't be found. Error occurred with RFR_Tools installation.
`

**I understand my case is unusual - sorry to be a bother.**

# DETAILS OF MY SYSTEM
OS:  Ubuntu 20.04.2 LTS 64-bit Server
Kernal: Linux ROSbot 5.4.0-1036-raspi #39-Ubuntu SMP PREEMPT Wed May 12 17:37:51 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
Python3: Python 3.8.10
Python2: Python 2.7.18 - "python2 --version"
PIP/PIP3:
```
pi@ROSbot:~$ pip --version
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
pi@ROSbot:~$ pip3 --version
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
```
**NO PIP2**:
```
pi@ROSbot:~$ pip2

Command 'pip2' not found, did you mean:

  command 'pipx' from deb pipx (0.12.3.1-2ubuntu1)
  command 'pip3' from deb python3-pip (20.0.2-5ubuntu1.5)
  command 'nip2' from deb nip2 (8.7.0-1)
  command 'pip' from deb python3-pip (20.0.2-5ubuntu1.5)
```
GoPiGo3: (16-tick model) 
```
GoPiGo3 info:
Manufacturer    :  Dexter Industries
Board           :  GoPiGo3
Serial Number   :  56ECD67E5152415447202020FF192614
Hardware version:  3.x.x
Firmware version:  1.0.0
```

Tried with bypass rfrtools, and then full update minus gui:

# FAILED: Update Python package (only)

```
pi@ROSbot:~$ curl -kL dexterindustries.com/update_gopigo3| bash -s -- --bypass-rfrtools --no-dependencies

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   251  100   251    0     0   1394      0 --:--:-- --:--:-- --:--:--  1394
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 12113  100 12113    0     0  11092      0  0:00:01  0:00:01 --:--:-- 11092
  _____            _                                
 |  __ \          | |                               
 | |  | | _____  _| |_ ___ _ __                     
 | |  | |/ _ \ \/ / __/ _ \ '__|                    
 | |__| |  __/>  <| ||  __/ |                       
 |_____/ \___/_/\_\\__\___|_|          _            
 |_   _|         | |         | |      (_)           
   | |  _ __   __| |_   _ ___| |_ _ __ _  ___  ___  
   | | | '_ \ / _\ | | | / __| __| '__| |/ _ \/ __| 
  _| |_| | | | (_| | |_| \__ \ |_| |  | |  __/\__ \ 
 |_____|_| |_|\__,_|\__,_|___/\__|_|  |_|\___||___/ 
                                                    
                                                    
   ____       ____  _  ____      _____ 
  / ___| ___ |  _ \(_)/ ___| ___|___ / 
 | |  _ / _ \| |_) | | |  _ / _ \ |_ \ 
 | |_| | (_) |  __/| | |_| | (_) |__) |
[... 132 lines cut ...]
    23756  2021-05-31 15:48   di_i2c.py
     1700  2021-05-31 15:48   di_mutex.py
      303  2021-05-31 15:51   EGG-INFO/PKG-INFO
      372  2021-05-31 15:51   EGG-INFO/SOURCES.txt
        1  2021-05-31 15:51   EGG-INFO/dependency_links.txt
       46  2021-05-31 15:51   EGG-INFO/requires.txt
       60  2021-05-31 15:51   EGG-INFO/top_level.txt
        1  2021-05-31 15:51   EGG-INFO/zip-safe
     2812  2021-05-31 15:51   __pycache__/I2C_mutex.cpython-38.pyc
     6565  2021-05-31 15:51   __pycache__/auto_detect_robot.cpython-38.pyc
     4592  2021-05-31 15:51   __pycache__/auto_detect_rpi.cpython-38.pyc
    14282  2021-05-31 15:51   __pycache__/di_i2c.cpython-38.pyc
     1652  2021-05-31 15:51   __pycache__/di_mutex.cpython-38.pyc
---------                     -------
    74952                     16 files
```

# HOW GoPiGo3 and DI_Sensors WERE ORIGINALLY INSTALLED
```
==== SETUP GoPiGo3 and DI_Sensors Python3 eggs 
$ cd /home/pi/Dexter/GoPiGo3/Software/Python
$ sudo python3 setup.py install
$ cd /home/pi/Dexter/DI_Sensors/Python
$ sudo python3 setup.py install
```

CleoQc, 2021-07-18:
I removed pip2 (well I renamed it as pip for now although that will lead to issues on GoPiGo OS. I have to revisit this)

slowrunner, 2021-07-19:
Thanks @CleoQc but I still got some errors installing, so I found a way that works.  I think you should revert the change you did for me.

For anyone that might need to update GoPiGo3, DI_Sensors, and R4R_Tools(gets I2C_mutex):
```
=== UPDATE THAT WORKED
(!!! log in as pi  !!!)

- BACKUP ~/Dexter
cp -r ~/Dexter ~/Dexter.old
rm -rf ~/Dexter

- Get Dexter/GoPiGo3  (from master)
$ sudo git clone http://www.github.com/DexterInd/GoPiGo3.git /home/pi/Dexter/GoPiGo3

- Get Dexter/DI_Sensors
$ sudo git clone https://github.com/DexterInd/DI_Sensors.git /home/pi/Dexter/DI_Sensors

- Setup RFR_Tools (for I2C_mutex)
$ sudo git clone https://github.com/DexterInd/RFR_Tools.git /home/pi/Dexter/lib/Dexter/RFR_Tools
$ cd /home/pi/Dexter/lib/Dexter//RFR_Tools/miscellaneous/
$ sudo python3 setup.py install

- SETUP GoPiGo3  Python3 egg 
$ cd /home/pi/Dexter/GoPiGo3/Software/Python
$ sudo python3 setup.py install
...
Installed /usr/local/lib/python3.8/dist-packages/gopigo3-1.3.0-py3.8.egg

- Install DI_Sensors egg
$ cd /home/pi/Dexter/DI_Sensors/Python
$ sudo python3 setup.py install

- Get serial number file
$ wget -O .list_of_serial_numbers.pkl https://raw.githubusercontent.com/DexterInd/GoPiGo3/master/Install/list_of_serial_numbers.pkl

- Check your python is using the installed package
$ python3
Python 3.8.10 (default, Jun  2 2021, 10:49:15) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gopigo3
>>> print(gopigo3.__file__)
/usr/local/lib/python3.8/dist-packages/gopigo3-1.3.0-py3.8.egg/gopigo3.py

- Check what methods are in GoPiGo3 class:
$ python3
>>> import gopigo3
>>> for f in dir(gopigo3.GoPiGo3):
...   print(f)
... 

look for
_check_serial_number_for_16_ticks

- [OPTIONAL - Create default ~/Dexter/gpg3_config.json

>>> gpg3=gopigo3.GoPiGo3()
(ctrl-D to exit)

more ~/Dexter/gpg3_config.json

- [MY OPTION] - copy gpg_config.json
cp ~/rosbot-on-gopigo3/configs/tilde-Dexter.gpg3_config.json ~/Dexter/gpg3_config.json

```

closed 2021-07-19 by slowrunner