DexterInd/GoPiGo3 #320: Run GoPiGo3 w/ RHEL/CentOS/Fedora IoT?

goetzrieger, 2022-05-12:
Hi there:

Is it feasible or does anybody has experience running the GoPiGo3 with a non-Debian Linux distro like in $SUBJECT?

Background: We have a small fleet of GoPiGo3's (around 15) we used to run event workshops for our (Red Hat, that is) developer line of products. Until, well, Covid came along. So it's been 2.5 years and we would like to revive this, but would like to brush up our game:

- We'd like to use the robots more as "edge devices", again to show our product range in this area
- We plan to run some ML workloads (think road/line following) directly on the robots
- Thinking about running MicroShift (a IoT-device version of our Container Platform) on the small guys 

All of this requires one of the above distros running and probably a raspi 4 and something like a Google Coral TPU stick.

Before I start digging into the setup scripts and hitting roadblocks any help would be appreciated!

CleoQc, 2022-05-12:
Hello there and welcome back!
The GoPiGo3 can be run with minimal installation on other distros as long as you do some setup manually. 

Of the top of my head, you need to make sure SPI and I2C are active.  Lately the GoPiGo3 uses pigpiod which may not be available on other distros so you would need some changes in gopigo3.py

The import : https://github.com/DexterInd/GoPiGo3/blob/master/Software/Python/gopigo3.py#L30

And these setup lines
https://github.com/DexterInd/GoPiGo3/blob/master/Software/Python/gopigo3.py#L231-L235

And this install script:
https://github.com/DexterInd/GoPiGo3/blob/7652f5b8c1815b7e51d5bc568f347e6fd0d1efa4/Install/install.sh#L47-L51

You won't need any of the tools that are tailored to beginners and that saves a lot of hassle.

Instead of running the install script, you can go ahead and run

`python3 setup.py install `

I _think_ it will work...

CleoQc, 2022-05-12:
@slowrunner has been able to install it on Ubuntu, he may have some words of wisdom.

jharris1993, 2022-05-13:
Slowrunner, also known as "cyclicalobsessive" has done a considerable body of work on this exact subject - getting Dexter Industries GoPiGo software to work on other platforms.  Please check his work on the Dexter forums at [https://forum.dexterindustries.com](https://forum.dexterindustries.com)

One possible kicker:
Most everything we've done is using software based off of Debian/Ubuntu distributions and, (as far as I know), it has never been attempted on a non-Debian derived distribution.

Please go to the forums, post your question there too, and if you try this with any non-Debian based release, let us know what worked and what didn't.

Thanks!

slowrunner, 2022-05-13:
@goetzrieger These are my steps to install the basic GoPiGo3 API and supporting configuration on Ubuntu 20.0.4:
```
==== BEGIN GOPIGO3 SPECIFIC STEPS

===== Create a Pi user with a /home/pi default directory, /bin/bash login shell ===

$ sudo useradd -m pi
$ sudo passwd pi

$ sudo usermod -a -G ubuntu,adm,dialout,cdrom,floppy,sudo,audio,dip,video,plugdev,netdev,lxd pi
$ sudo groupadd gpio
$ sudo groupadd i2c
$ sudo groupadd spi
$ sudo usermod -a -G gpio,i2c,spi pi
$ sudo usermod -a -G gpio,i2c,spi ubuntu
$ sudo usermod -a -G pi ubuntu

Check with:
$ groups pi
$ groups ubuntu

=== CONFIGURE PASSWORD-LESS SUDO

sudo nano /etc/sudoers

make sudo group look like:
# Allow members of group sudo to execute any command
%sudo	ALL=(ALL:ALL) NOPASSWD: ALL

=== CONFIGURE PI DEFAULT SHELL TO BASH

$ sudo chsh -s /bin/bash pi

=== ALLOW non-root SPI and GPIO  (Temporarily - will make permanent later)

$ sudo chgrp spi /dev/spidev0.0
$ sudo chgrp spi /dev/spidev0.1

$ sudo chmod 660 /dev/spidev0.0
$ sudo chmod 660 /dev/spidev0.1

$ sudo chgrp gpio /sys/class/gpio/export
$ sudo chgrp gpio /sys/class/gpio/unexport

$ sudo chmod 770 /sys/class/gpio/export
$ sudo chmod 770 /sys/class/gpio/unexport

=== [SKIP] === Enable spi, i2c without raspi-config

6/3/21: system-boot/sysconfig.txt now has these configured

enable_uart=1
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

==== GOPIGO3 LIBRARIES 
$ logout
$ ssh pi@x.x.x.x
$ sudo git clone -b install_on_ubuntu http://www.github.com/DexterInd/GoPiGo3.git /home/pi/Dexter/GoPiGo3

$ sudo curl -kL dexterindustries.com/update_tools | bash -s -- --system-wide --use-python3-exe-too --install-deb-debs --install-python-package
[sudo] password for pi:  (enter login pswd for pi)

$ sudo apt-get install --no-install-recommends python3-pip python3-numpy python3-curtsies

=== DI_SENSORS LIBRARY

$ sudo git clone https://github.com/DexterInd/DI_Sensors.git /home/pi/Dexter/DI_Sensors

[... 206 lines cut ...]

=== ESPEAK-NG 
  - Not installing python3-espeak or speech-dispatcher-espeak-ng (installs pulseaudio)
$ sudo apt-get install -y espeak-ng
$ espeak-ng "Hello"
  - if want python interface to espeak-ng:
    $ pip3 install py-espeak-ng

==== FULL POWEROFF FOR CLEAN TEST
$ sudo shutdown -h now

Boot up in GoPiGo3

$ ssh pi@X.X.X.X

=== Test GoPiGo Functions
$ cd ~
$ python3 Dexter/GoPiGo3/Software/Python/Examples/Read_Info.py
$ python3 Dexter/GoPiGo3/Software/Python/Examples/LED.py
$ python3 Dexter/GoPiGo3/Software/Python/Examples/easy_Distance_Sensor.py
$ python3 Dexter/DI_Sensors/Python/Examples/IMUSensor.py

$ python3 Dexter/GoPiGo3/Software/Python/Examples/Servo.py
$ python3 Dexter/GoPiGo3/Software/Python/Examples/Motor_Turn.py
```

jharris1993, 2022-05-14:
One thing that is implicit in all of this is that ***there is a hard dependency on the "pi" user*** as the primary, (default), user.
(_i.e._ When you create your initial username for the system, it should be "pi".)

This is a known "issue" based on the fact that the GoPiGo libraries were originally designed to be installed over a stock Raspbian install where the default username was "pi".  This is known and there are plans to fix this - I don't know if or when.

The "pi" user should also be in the sudoers file and in the wheel group to make sure it can do whatever you need it to do.

References to raspi-config do not apply to non Raspbian based systems.

goetzrieger, 2022-05-14:
Wow, good to see there is a really active community around! Even after this Covid-induced time of very little in-person events. But I‘m really eager going back running Robot Hackathons now. And Nicole, I very well remember how you helped out with faulty boards on short notice to rescue one of our event in 2019.

Okay. From what I understand there is an „old-style“ sysfs interface to GPIO that is deprecated since Kernel 4.18 or so and the new /dev/gpiochip* interface. Most modern distros are starting to use the new interface, notably Fedora-based ones. https://embeddedbits.org/new-linux-kernel-gpio-user-space-interface/

So on Fedora the old sysfs-style interface is not enabled in the kernel and pigpiod is not available.

Nicole mentioned above Gopigo3 Python libs are using pigpiod now which I understand is using the sysfs interface. Does that mean there is a lot of work in porting the gopigo3 libs/API to the new interface? The library is called libpigpiod and has Python bindings, too.

I’m being a bit verbose because I‘m pretty new to this stuff and this would put a stop on my experiment already.

goetzrieger, 2022-05-14:
@jharris1993 Fedora is kind of the upstream community project to our Red Hat Enterprise Linux. We are moving fast into the „Edge“ use cases so Fedora and in the newest version RHEL are able to run on ARM as found in lots of IoT devices. https://docs.fedoraproject.org/en-US/quick-docs/raspberry-pi/

My main target for our GopiGo’s is Fedora IoT which is running on Raspi quite well and is a great foundation for deploying containerized applications. https://getfedora.org/en/iot/

This would e.g. allow us to run the Gopigo libs in a container and maybe something like a Tensorflow-lite based image recognition application in another one on the robot.

goetzrieger, 2022-05-14:
But first I‘ve got to get the robot running with Fedora.