DexterInd/GoPiGo3 #337: Readme.md still references Raspbian for Robots as if it were still fully supported

jharris1993, 2024-07-17:
Issue:
The readme.md file still references Raspbian for Robots as if it were a valid and supported operating system - which is not true.

Recommendation:
The reference to Raspbian for Robots, (assuming it is still available for download), should be clearly marked as depreciated and unsupported. (except maybe on the forums), and is only suitable for older robots and/or Grove Pi, Brick Pi, etc.

Since I am not sure how to word this and remain in compliance with your marketing department, I have decided not to create a corresponding pull request.

slowrunner, 2024-07-17:
Perhaps more egregious than R4R not mentioning "deprecated - not a supported configuration", are:
- [Quick Install](https://github.com/DexterInd/GoPiGo3#quick-install)
- [Virtual Environment Installation](https://github.com/DexterInd/GoPiGo3#virtual-environment-installation)
- [Installation](https://github.com/DexterInd/GoPiGo3#installation)

Once or twice a month we get folks on the forum asking why their install for GrovePi or GoPiGo3 did not work, which ultimately ends up they are not using the 32-bit Legacy PiOS.

[Lets "crowd source" a new README.md proposal](https://forum.dexterindustries.com/t/help-update-the-gopigo3-readme/9861?u=cyclicalobsessive) and when all are happy with it, one of us can submit it as a pull request.

jharris1993, 2024-07-18:
>[Lets "crowd source" a new README.md proposal](https://forum.dexterindustries.com/t/help-update-the-gopigo3-readme/9861?u=cyclicalobsessive) and when all are happy with it, one of us can submit it as a pull request.

Entirely agree.

There's actually two ways to do this:
1.  Continue discussing on the forums as noted above.
2. Create a pull request with updated content and discuss it/modify it here.
3.

closed 2026-05-29 by CleoQc