DexterInd/GoPiGo3 #339: update_tools fails with a script_tools installer error: "the user must be pi" even though the user IS pi

jharris1993, 2024-09-19:
When running
`curl -kL dexterindustries.com/update_tools | sudo bash`
it fails with the following error:
`script_tools installer script must be run with "pi" user. Exiting.`

Viz.:
```
pi@raspberrypi:~ $ curl -kL dexterindustries.com/update_tools | sudo bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   249  100   249    0     0    602      0 --:--:-- --:--:-- --:--:--   601
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100  2145  100  2145    0     0   1485      0  0:00:01  0:00:01 --:--:--  1485
script_tools installer script must be run with "pi" user. Exiting.
pi@raspberrypi:~ $ whoami
pi
pi@raspberrypi:~ $
```

Thanks!
Jim

slowrunner, 2024-09-19:
Jim, piping it to "sudo bash" is running bash as "root"

jharris1993, 2024-09-20:
Absolutely.

However that was the published way.

CleoQc, 2024-10-29:
have you tried 
`curl -kL dexterindustries.com/update_tools | bash `
?

slowrunner, 2024-10-29:
> However that was the published way.

Where are you referring to?  This is a doc error?